unitreerobotics/unitree_rl_lab #7: Runtime Error in Training

sssn-tech, 2025-07-08:
Hi team,

I've encountered an issue during training with the following command:
```
python scripts/rsl_rl/train.py --headless --task=Unitree-H1-Velocity
```

Everything runs fine until a large number of timesteps is reached. Here's the relevant log:
```
--------------------------------------------------------------------------------
                   Total timesteps: 4379934720
                    Iteration time: 0.99s
                      Time elapsed: 12:14:33
                               ETA: 01:29:47

Error executing job with overrides: []
Traceback (most recent call last):
  File "/home/ecs-user/IsaacLab/source/isaaclab_tasks/isaaclab_tasks/utils/hydra.py", line 101, in hydra_main
    func(env_cfg, agent_cfg, *args, **kwargs)
  File "/home/ecs-user/working-labs/unitree_rl_lab/scripts/rsl_rl/train.py", line 182, in main
    runner.learn(num_learning_iterations=agent_cfg.max_iterations, init_at_random_ep_len=True)
  File "/home/ecs-user/miniconda3/envs/is/lib/python3.10/site-packages/rsl_rl/runners/on_policy_runner.py", line 262, in learn
    loss_dict = self.alg.update()
  File "/home/ecs-user/miniconda3/envs/is/lib/python3.10/site-packages/rsl_rl/algorithms/ppo.py", line 260, in update
    self.policy.act(obs_batch, masks=masks_batch, hidden_states=hid_states_batch[0])
  File "/home/ecs-user/miniconda3/envs/is/lib/python3.10/site-packages/rsl_rl/modules/actor_critic.py", line 122, in act
    return self.distribution.sample()
  File "/home/ecs-user/miniconda3/envs/is/lib/python3.10/site-packages/torch/distributions/normal.py", line 73, in sample
    return torch.normal(self.loc.expand(shape), self.scale.expand(shape))
RuntimeError: normal expects all elements of std >= 0.0
```

I also encountered the same error while training the `Unitree-G1-29dof-Velocity` task. In that case, the crash occurred after about 3 hours of training, with an ETA still around 15 hours.

So far I’ve only tested these two tasks, and both produced the same runtime error. It appears to be caused by invalid standard deviation values in the policy distribution.

Here are the versions I'm using:

```
(is) ecs-user@iZuf6g6vii5ci3ikxssy9lZ:~/IsaacLab$ cat VERSION
2.1.0
(is) ecs-user@iZuf6g6vii5ci3ikxssy9lZ:~/IsaacLab$ pip show isaacsim
Name: isaacsim
Version: 4.5.0.0
Summary: A metapackage to manage the Isaac Sim installation
Home-page: https://developer.nvidia.com/isaac-sim
Author: Isaac Sim team
Author-email:
License: NVIDIA Proprietary Software
Location: /home/ecs-user/miniconda3/envs/is/lib/python3.10/site-packages
Requires: isaacsim-kernel
Required-by:
```

Let me know if more information is needed — I’d be happy to share logs or configs.
I hope informations above are helpful. Any suggestions or workarounds would be much appreciated. Thanks in advance!

Agnel-Wang, 2025-07-08:
Add `bad_orientation` to terminations will fix it 
see commit e72a072c4db68b79a816b3b89bf3b6a0525d113e

closed 2025-07-09 by sssn-tech

HonzaKubita, 2026-04-25:
This still happens even after the patch introduced in e72a072c4db68b79a816b3b89bf3b6a0525d113e
I have ran the training and after couple of hours I got the same error with `bad_orientation` in the `TerminationsCfg` present