unitreerobotics/unitree_rl_lab #24: Trainig fails due to std >=0

filippottaviani, 2025-07-29:
After 4 hours of training ( running train.py script) I encountered this error:

Error executing job with overrides: []
Traceback (most recent call last):
  File "/home/filippo/IsaacLab/source/isaaclab_tasks/isaaclab_tasks/utils/hydra.py", line 101, in hydra_main
    func(env_cfg, agent_cfg, *args, **kwargs)
  File "/home/filippo/unitree_rl_lab/scripts/rsl_rl/train.py", line 182, in main
    runner.learn(num_learning_iterations=agent_cfg.max_iterations, init_at_random_ep_len=True)
  File "/home/filippo/env_isaaclab/lib/python3.10/site-packages/rsl_rl/runners/on_policy_runner.py", line 262, in learn
    loss_dict = self.alg.update()
  File "/home/filippo/env_isaaclab/lib/python3.10/site-packages/rsl_rl/algorithms/ppo.py", line 260, in update
    self.policy.act(obs_batch, masks=masks_batch, hidden_states=hid_states_batch[0])
  File "/home/filippo/env_isaaclab/lib/python3.10/site-packages/rsl_rl/modules/actor_critic.py", line 124, in act
    return self.distribution.sample()
  File "/home/filippo/env_isaaclab/lib/python3.10/site-packages/torch/distributions/normal.py", line 73, in sample
    return torch.normal(self.loc.expand(shape), self.scale.expand(shape))
RuntimeError: normal expects all elements of std >= 0.0

I read the reply to  the related "resolved" issue but the "bad_orientation termination doesn't fix the error at all. I didn't modify the code except for the USD model path declaration.

zhenwang-civil, 2025-07-30:
I modified the value of limit_angle from 0.8 to 0.7 in the following line:
bad_orientation = DoneTerm(func=mdp.bad_orientation, params={"limit_angle": 0.8})
located at source/unitree_rl_lab/unitree_rl_lab/tasks/locomotion/robots/g1/29dof/velocity_env_cfg.py.
After this change, the issue no longer occurs. I hope this helps you as well.

filippottaviani, 2025-07-30:
Thanks for the hint but using bad_orientation like that terminates the episode too quickly. This happens because of the  randomization of the robot pose from the related event.

filippottaviani, 2025-11-28:
> Changing the limit value in this line bad_orientation = DoneTerm(func=mdp.bad_orientation, params={"limit_angle": 0.8}) from 0.8 to 0.7, I tried 0.6, it still does not solve the problem for me. Anyone here has the same situation as me and managed to solved it?

I have the same issue and bad_orientation doesn't solve it. I also read the error traceback and I can't figure out how using that termination could help.

filippottaviani, 2025-12-10:
> > > Changing the limit value in this line bad_orientation = DoneTerm(func=mdp.bad_orientation, params={"limit_angle": 0.8}) from 0.8 to 0.7, I tried 0.6, it still does not solve the problem for me. Anyone here has the same situation as me and managed to solved it?
> > 
> > 
> > I have the same issue and bad_orientation doesn't solve it. I also read the error traceback and I can't figure out how using that termination could help.
> 
> what gpu and driver are you using? i am using rtx 3060, driver 580

RTX 4090 driver 580

filippottaviani, 2025-12-18:
> This is because RSL_RL received a NaN or Inf for policy input. I think you are probably training on uneven terrain, sometimes the raycaster may penetrate throught the terrain mesh, resulting Inf or NaN output. The abnormal value can also be fed through reward and obs "base_height" related term.
> 
> I add extra safety check for base height, impl a new base height reward below:
> 
> def base_height_relative_l2(
>     env: ManagerBasedRLEnv,
>     target_height: float,
>     asset_cfg: SceneEntityCfg = SceneEntityCfg("robot"),
>     sensor_cfg: SceneEntityCfg | None = None,
> ) -> torch.Tensor:
>     # Extract the asset
>     asset: RigidObject = env.scene[asset_cfg.name]
> 
>     if sensor_cfg is not None:
>         from isaaclab.sensors import RayCaster
>         sensor: RayCaster = env.scene[sensor_cfg.name]
>         relative_height = asset.data.root_pos_w[:, 2] - torch.nanmean(sensor.data.ray_hits_w[..., 2], dim=1)
>         reward = torch.square(relative_height - target_height)
>         # if any rew is NaN/inf, return 0 for these idx
>         reward = torch.where(torch.isnan(reward) | torch.isinf(reward), torch.zeros_like(reward), reward)
> 
>         return torch.clamp(reward, 0.0, 100.0)
>     else:
>         # Flat terrain: use absolute world height
>         reward = torch.square(asset.data.root_pos_w[:, 2] - target_height)
>         reward = torch.where(torch.isnan(reward) | torch.isinf(reward), torch.zeros_like(reward), reward)
>         return torch.clamp(reward, 0.0, 100.0)
> which would be helpful.

Thanks for the reply but I'm using no raycaster