unitreerobotics/unitree_rl_lab #31: deploy sim2sim result not really working

thomasschichl, 2025-08-12:
Dear community,

I am facing an issue with the G1 29dof robot. When using the unitree_rl_lab repo and following all its instructions I am able to train and play a G1 robot.
Just for testing I used the prepared enviroment configuration provided by the repo.
After training the robot and testing the robot in isaac sim the robot is walking more or less smothly. However, when deploying the pt file in mujoco (using simulate_python and not simulte with c++) the robot is not really controllable. The robot drifts to the left side and when using the joystick to control it, it is nearly impossible.
So I thought, ok I will try the prepared policy file from unitree in the v0 folder. So I changed the path in the yaml file restartet mujoco and starte the g1_ctrl- sam the thing. Robot is hardly to control. However, it is a bit better to control. So as the sim2sim is not really providing a good result I do not really dare to try it on the real robot...
So again
1 used the most up to date unitree_rl_lab repo
2 installed isaac sim and isaac lab according to docu
3 trained the unitree g1 robot using the prepared enviroment by unitree
4 played the result to get the pt file
5 opened unitree_mujoco with trhe g1 and joystick support
6 started the g1_ctrl
7 communication works- but robot is very instable and when releasing the elastic band the robot is hard to control
8 tried the same procedure with the prepared pt file from untiree- little better but still hard to control

I tried training a pt file with more iterations but same result. The robot is always shaking its arms with high frequency and low amplitude and the robot is always drifting to the left side and is always shaking very fast...

Did you use the pt file prvided in the repo v0 folder in a sim2sim test and did you test it on the real robot
Did you test the robot enviroment provided in the [scripts folder for the ](https://github.com/unitreerobotics/unitree_rl_lab/blob/main/source/unitree_rl_lab/unitree_rl_lab/tasks/locomotion/robots/g1/29dof/velocity_env_cfg.py) folder?
Is there something to take in consideration before starting the training?
I was wondering about the velocity ranges in the commands section- seem a little odd to me...

Anyone able to train a proper pt file?
We have already trained a lot of pt files using the enviorments provided by IsaacLab and they looked better in the. However, they are using completely different config files for the robot and parameters in the enviroments...

Kind regards.

thomasschichl, 2025-08-21:
> ##### Position Command for robot
> 
> A simple and effective way is to directly make the velocity command into position by pd. which dont change the base `stable` velocity policy
> ##### manipulation
> 
> The main method for manipulation is still VLA, there is no clear and effective way to use RL. If you want to use G1, you first need to develop a whole body control policy, or just use velocity to control legs, and supports teleoperation for the upper body

Dear Agnel-Wang,

Thanky ou for your response. I really appreciate your input. However, I am not quite sure if I understand your reamrks correctly.

What do you exaclty mean with "directly make the velocity command into position by pd". Or do you just mean the High Level command set_velocity(x,y,o,t)
Are you using the high level commands for the creation of taks as well?
Is it possible for users to run a trained file, just like the v0, and set paramters like within the v0 the analog value of the controller as the velocity. Obviously it is possible with the g1_ctrl file- but is it possible for me as well, if I want to set it using a python script for example.

As for manipulation. We are using the unitree_sim and xr_teleport repos as well for pick an place tasks to record episodes and use de IL_unitree repo to train a model. However, I was interested in trying to use the UniformPoseCommandCfg to train a model for reaching a point with the hand of the robot. Afterwards I wanted to track a object with the intel realsense and provide so the position for the robot to reach...- I am aware that the intel realsense is not very accurate and the robot itself is not really stiff (compared to a industrial robot arm). However I am in terested which of the methods is more accurate- RL or IL.
If the RL method would be accurate enough then it would be more flexible as through the camera it is pretty easy to identify multiple different objects compared to IL.

Are there plans to release a UniformPoseCommandCfg example for the G1 in the near future?

marco-ambrogi, 2025-08-25:
> > Hello! May I ask why when I'm running the ./g1_ctrl file during sim2sim, it keeps showing: (env_isaaclab) GeekPro-17IRR:~/unitree_rl_lab-main/deploy/robots/g1/build$ ./g1_ctrl --- Unitree Robotics --- G1-29dof Controller [2025-08-20 13:51:24.018] [info] Waiting for connection to robot... [2025-08-20 13:51:26.019] [warning] Waiting for connection rt/lowstate, and then there's no response for a long time? Am I missing some key steps?
> 
> use ./g1_ctrl --network lo

Hello, I have same problem also using ./g1_ctrl --network lo, do you have any ideas to go on?

thomasschichl, 2025-08-26:
> **h1_locomotion.py** is just an example to call policy in isaaclab. However, It's not recommended to deploy a locomotion policy by python.
> 
> The simple way in unitree_rl_lab is like this:
> 
> #include <unitree/idl/ros2/std_msgs/String_.hpp>
> #include <nlohmann/json.hpp>
> 
> namespace isaaclab
> {
> 
> REGISTER_OBSERVATION(other_command)
> {
>     static auto sub = std::make_shared<unitree::robot::SubscriptionBase<std_msgs::msg::dds_::String_>>("rt/topic_name");
>     nlohmann::json j = nlohmann::json::parse(sub->msg_.data());
>     std::vector<float> velocit_commands = j["data"].get<std::vector<float>>();
>     return velocit_commands;
> }
> 
> you just need to publish your command as _std_msgs::string_(or other idl) type to a topic and rename your _velocity_commands_ obs in deploy.yaml

Dear Agnel-Wang,

Thats just perfext. I have already noticed in some examples, that in the yaml files instead of values there are variables. Howeever, I was not able to find anything in the isaaclab documentation on how to address this variables. This is a great help!
Really appreciated. I will give it a try the next days.

One last basic question. In the env.yaml files within the training results in the logs folder there are ini values for the stiffness and damping. When the training is done, will there only be the angle identfied for lets say locomotion or will the values for the stiffness and damping also be identifed by the training. I mean in the rewards, observations etc and commands there is nothing about stiffness or damping. So are these values also "trained" or is it just about the values of the angles?

Really appreciate your help and I think the hint with the deploy.yaml will helkp us a lot!

kind regards!

Edwardle06, 2025-08-27:
> Dear community,
> 
> I am facing an issue with the G1 29dof robot. When using the unitree_rl_lab repo and following all its instructions I am able to train and play a G1 robot. Just for testing I used the prepared enviroment configuration provided by the repo. After training the robot and testing the robot in isaac sim the robot is walking more or less smothly. However, when deploying the pt file in mujoco (using simulate_python and not simulte with c++) the robot is not really controllable. The robot drifts to the left side and when using the joystick to control it, it is nearly impossible. So I thought, ok I will try the prepared policy file from unitree in the v0 folder. So I changed the path in the yaml file restartet mujoco and starte the g1_ctrl- sam the thing. Robot is hardly to control. However, it is a bit better to control. So as the sim2sim is not really providing a good result I do not really dare to try it on the real robot... So again 1 used the most up to date unitree_rl_lab repo 2 installed isaac sim and isaac lab according to docu 3 trained the unitree g1 robot using the prepared enviroment by unitree 4 played the result to get the pt file 5 opened unitree_mujoco with trhe g1 and joystick support 6 started the g1_ctrl 7 communication works- but robot is very instable and when releasing the elastic band the robot is hard to control 8 tried the same procedure with the prepared pt file from untiree- little better but still hard to control
> 
> I tried training a pt file with more iterations but same result. The robot is always shaking its arms with high frequency and low amplitude and the robot is always drifting to the left side and is always shaking very fast...
> 
> Did you use the pt file prvided in the repo v0 folder in a sim2sim test and did you test it on the real robot Did you test the robot enviroment provided in the [scripts folder for the ](https://github.com/unitreerobotics/unitree_rl_lab/blob/main/source/unitree_rl_lab/unitree_rl_lab/tasks/locomotion/robots/g1/29dof/velocity_env_cfg.py) folder? Is there something to take in consideration before starting the training? I was wondering about the velocity ranges in the commands section- seem a little odd to me...
> 
> Anyone able to train a proper pt file? We have already trained a lot of pt files using the enviorments provided by IsaacLab and they looked better in the. However, they are using completely different config files for the robot and parameters in the enviroments...
> 
> Here is a video: Using the pt file trained by me based on the preapred enviroment von unitree At first the robot stands steady After pressing L2+arrow up on the controller the robot starts jittering After pressinf R1+x on th econtroller the robot starts moving wildly After adjusting the robots position with 7 and 8 so that it stands proper on the floor the robot stands a bit better After releasing the elastic band with 9 the robot drifts a little bit When using the left and right joystick the robto is hard to control
> 
>  untitled.mp4 
> Kind regards.
Hello，I'm facing the same issue as you. When I press R1+X, the robot starts swinging violently. Even after modifying the armature, there’s still minor jitter—it won’t stay completely still. How did you eventually solve this? Additionally, after pressing R1+X, no matter how I move the joystick, the robot only moves in place and fails to execute the policy. Looking forward to your reply.

Agnel-Wang, 2025-08-27:
> > **h1_locomotion.py** is just an example to call policy in isaaclab. However, It's not recommended to deploy a locomotion policy by python.
> > The simple way in unitree_rl_lab is like this:
> > #include <unitree/idl/ros2/std_msgs/String_.hpp>
> > #include <nlohmann/json.hpp>
> > namespace isaaclab
> > {
> > REGISTER_OBSERVATION(other_command)
> > {
> > static auto sub = std::make_shared<unitree::robot::SubscriptionBase<std_msgs::msg::dds_::String_>>("rt/topic_name");
> > nlohmann::json j = nlohmann::json::parse(sub->msg_.data());
> > std::vector velocit_commands = j["data"].get<std::vector>();
> > return velocit_commands;
> > }
> > you just need to publish your command as _std_msgs::string_(or other idl) type to a topic and rename your _velocity_commands_ obs in deploy.yaml
> 
> Dear Agnel-Wang,
> 
> Thats just perfext. I have already noticed in some examples, that in the yaml files instead of values there are variables. Howeever, I was not able to find anything in the isaaclab documentation on how to address this variables. This is a great help! Really appreciated. I will give it a try the next days.
> 
> One last basic question. In the env.yaml files within the training results in the logs folder there are ini values for the stiffness and damping. When the training is done, will there only be the angle identfied for lets say locomotion or will the values for the stiffness and damping also be identifed by the training. I mean in the rewards, observations etc and commands there is nothing about stiffness or damping. So are these values also "trained" or is it just about the values of the angles?
> 
> Really appreciate your help and I think the hint with the deploy.yaml will helkp us a lot!
> 
> kind regards!

The thing about kpkd in training is in the domain_randomize (randomize motor gain), but when deploying, you just need the default settings (see deploy.yaml).
deploy.yaml is automatically generated.
It contains all the parameters related to deployment.
For the policy, the output is only the joint position action.

closed 2025-10-17 by Agnel-Wang