unitreerobotics/unitree_rl_lab #34: Why can the robot walk normally in Isaac Sim, but exhibit severe foot shaking in MuJoCo?

Edwardle06, 2025-08-21:
Hello! Have you ever trained a policy for over 20,000 iterations? I’d like to ask about a cross-platform testing issue: The policy enables stable walking in Isaac Sim, but when transferred to MuJoCo, the robot’s foots exhibit severe shaking. Could this be due to insufficient training iterations (e.g., needing to increase to 50,000)? Or might there be other underlying problems? I’d appreciate your insights.

Agnel-Wang, 2025-08-22:
Which robot and which type of shaking are you referring to?
Currently only G1-29dof provides pre-trained models

Performance is good in IsaacSim (the training environment) but degrades significantly in Mujoco (the deployment or real-world environment). This is a common sim-to-real challenge faced by everyone.
A standard approach for this issue is domain randomization.

Edwardle06, 2025-08-25:
Hello, thank you very much for your reply! I am not using a robot from Unitree, but a bipedal robot from another company. I would like to borrow your open-source project to enable my robot to train and run in Isaac Sim. Based on your suggestion, should I check the domain randomization part?

Agnel-Wang, 2025-08-25:
Yes, but there are many reasons for this problem; 
motor settings in mujoco; action_rate/joint_effort or other rewards; 
this can only be solved by yourself

Agnel-Wang, 2025-09-01:
disable [gait](https://github.com/unitreerobotics/unitree_rl_lab/blob/main/source/unitree_rl_lab/unitree_rl_lab/tasks/locomotion/mdp/rewards.py#L200) when no cmd