unitreerobotics/unitree_rl_lab #44: G1 Robot Unresponsive in Velocity Mode — Sim2Real Issue

zhenwang-civil, 2025-09-01:
I am currently facing a sim2real issue with the Unitree G1 robot. The steps I followed are:
1. Pressed L2 + R2 on the controller to enter debug mode.
2. In the terminal, ran: ./g1_ctrl --network enx6c1ff727580d
3. Using the official Unitree controller, pressed [L2 + Up] to enter FixStand mode, then [R1 + X] to switch to Velocity mode.
Observed behavior:
1. The G1 robot does not respond at all.
2. It does not enter a stable standing state.
3. Moving the joystick to issue velocity commands has no effect.
What could be the reason that the G1 robot is not responding, even though the connection appears to be normal? Could this be related to sim2real discrepancies or configuration issues?
wz@robot:~/unitree/unitree_rl_lab/deploy/robots/g1/build$ ./g1_ctrl --network enx6c1ff727580d --- Unitree Robotics --- G1-29dof Controller [2025-09-01 16:39:14.829] [info] Waiting for connection to robot... [2025-09-01 16:39:15.029] [info] Connected to robot. [2025-09-01 16:39:15.030] [info] FSM: Start Passive [2025-09-01 16:39:15.030] [info] Initializing State_Velocity... [2025-09-01 16:39:15.030] [info] Policy directory: /home/wz/unitree/unitree_rl_lab/deploy/robots/g1/../../../logs/rsl_rl/unitree_g1_29dof_velocity/2025-07-29_23-38-41 Press [L2 + Up] to enter FixStand mode. And then press [R1 + X] to start controlling the robot. [2025-09-01 16:39:25.116] [info] FSM: Change state from Passive to FixStand 8 [2025-09-01 16:39:39.675] [info] FSM: Change state from FixStand to Velocity

JaceLiu, 2025-09-04:
我也遇到了这个问题

JaceLiu, 2025-09-05:
我用的假手

JaceLiu, 2025-09-05:
这个问题我今天有了一定的进展，因为前几天代码有更新，我就重新把代码clone下来跑了一遍，需要注意的点是unitree_sdk2需要构建两次，一次是构建deploy时，cmake .. -DBUILD_EXAMPLES=OFF，一次是构建unitree_mujoco时，cmake .. -DCMAKE_INSTALL_PREFIX=/opt/unitree_robotics，这样就可以正常sim2real了，你如果用的是g1，可能会出现Unmatched robot type.的问题，出现这个问题的原因是我们的g1是23dof的，如果你是29dof的，应该就没问题了

Agnel-Wang, 2025-09-08:
> 这个问题我今天有了一定的进展，因为前几天代码有更新，我就重新把代码clone下来跑了一遍，需要注意的点是unitree_sdk2需要构建两次，一次是构建deploy时，cmake .. -DBUILD_EXAMPLES=OFF，一次是构建unitree_mujoco时，cmake .. -DCMAKE_INSTALL_PREFIX=/opt/unitree_robotics，这样就可以正常sim2real了，你如果用的是g1，可能会出现Unmatched robot type.的问题，出现这个问题的原因是我们的g1是23dof的，如果你是29dof的，应该就没问题了

啊是，unitree_mujoco现在的cmake是通过find(unitree_sdk2)来配置的，unitree_rl_lab里面是直接没配，用的/usr/local里面，这是开发人员的个人习惯问题，说不上对错，虽然确实有冲突。

Agnel-Wang, 2025-10-17:
Motors mode had added in the config.yaml.